openwisp/openwisp-config #15: New feature: allow url parameter to be a list

nemesisdesign, 2016-02-22:
A list would be very handy in order to allow more URLs, in case one fails.

sheerun, 2016-05-19:
I think this is rather load balancer responsibility to handle failover

BlancLoup, 2018-05-02:
I've forked openwisp-config to https://github.com/BlancLoup/openwisp-config
I've added redundant urls with backward compatibility, but have some issue to resolve:
Only way to check url connection is curl to "curl -Is --connect-timeout 10 -k -X POST https://server/controller/register/" that returns header "X-Openwisp-Controller: true". It's hack, not solution. Perfect: url something like "https://server/controller/heartbeat/" on controller's side.
NB: i have not tested yet this modification

nemesisdesign, 2018-05-03:
@BlancLoup could you open a pull request marked as **WIP** (work in progress) please? That way it will be easier to review and track.

Remember to reference this issue by adding `#15` in the description of the pull request.