openwisp/openwisp-config #19: Add smarter configuration merge mechanism

nemesisdesign, 2016-04-20:
We need a smarter configuration merging mechanism that does the following:

* does not cause issue #17 
* separates the remote configuration from the local one
* merges the remote configuration on the local one by default (with the optional possibility to overwrite instead of merging)

sheerun, 2016-05-19:
I think overwirte instead of merging is better default. Plus make merging configurable opt-in

nemesisdesign, 2016-05-20:
@sheerun there are several challenges in this task, if you overwrite by default you will have to keep a lot of boilerplate configuration in the controller, and implementing merge is not easy because of anonymous UCI configuration sections that cannot be merged, trying to merge two anonymous configsurations will result in two duplicated configurations

closed 2016-06-22 by nemesisdesign