openwisp/openwisp-config #26: run a command when openwisp fetch the configuration

gabri94, 2017-02-28:
In order to use Openwisp with LibreMesh, we need a way to call the command "lime-config" after the configuration has been fetched, but before the networking get reloaded.

I propose to add a parameter to specify a generic command in the config file.

nemesisdesign, 2017-02-28:
Question: does lime-config work with a single configuration file or multiple files?

Take a look at this script that is used internally by openwisp-config:

https://github.com/openwisp/openwisp-config/blob/master/openwisp-config/files/sbin/openwisp-reload-config

It's a modified version of OpenWRT's reload-config, which also explicitly calls /etc/init.d/<service> reload if necessary.

That's a way of doing it which wouldn't need to do new developments.

An alternative would be to add some kind of customisable hook to call when the configuration is applied, which would also be interesting.

G10h4ck, 2017-03-01:
lime-config read a couple of config files and write almost everything else

nemesisdesign, 2017-03-02:
@G10h4ck thanks for the info, could you add the link to the source code of `lime-config` please?

G10h4ck, 2017-03-02:
https://github.com/libremesh/lime-packages/blob/develop/packages/lime-system/files/usr/bin/lime-config

The modules are loaded from here

https://github.com/libremesh/lime-packages/tree/develop/packages/lime-system/files/usr/lib/lua/lime

nemesisdesign, 2017-03-03:
I guess the best thing is to add a generic hook to call `lime-config` as suggested by @gabri94. Stay tuned.

nemesisdesign, 2017-03-03:
Thx @gabri94 and @G10h4ck, see 033d69a. I will issue a new release later.

closed 2017-03-03 by nemesisdesign

nemesisdesign, 2017-03-03:
Released [openwisp-config 0.4.5](https://github.com/openwisp/openwisp-config/releases/tag/0.4.5).

commit 033d69ae (2017-03-03): Added pre-reload-hook #26

This change adds a pre-reload-hook mechanism to facilitate using
auto-configuration algorithms like the one used in libre-mesh.org

Closes #26