openwisp/openwisp-config #27: --capath is not supported on mbedtls and cyassl on LEDE

nemesisdesign, 2017-03-02:
See also:

* [lede forum: –capath not supported by libcurl with mbedTLS support](https://forum.lede-project.org/t/capath-not-supported-by-libcurl-with-mbedtls-support/947/8)
* [lede-dev: curl --capath not supported on mbedtls but was supported on polarssl](http://lists.infradead.org/pipermail/lede-dev/2017-March/006432.html)

I don't really know what to do with this. I hope I will get some feedback from the LEDE forum or mailing list.

nemesisdesign, 2017-03-03:
Created a task in the LEDE bug tracker: https://bugs.lede-project.org/index.php?do=details&task_id=589

nemesisdesign, 2017-03-07:
I found out a possible solution.

The packages for openssl and polarssl must depend on `ca-certificates` while the packages for mbedtls and wolfssl (formerly cyassl) must depend on `ca-bundle`.

the `--capath` argument must be supplied only if specified explicitly, while a `--cabundle` should be added for consistency with a similar behaviour (should be passed to curl only when explicitly defined).

nemesisdesign, 2017-03-16:
Unfortunately, the `ca-bundle` package was introduced only recently and depending on it would be a change not backward compatible with older OpenWRT versions.

I'm going to try an alternative solution which involves generating the certificate bundle file directly in the Makefile of *openwisp-config*.

nemesisdesign, 2017-03-16:
The problem seems fixed. If anyone finds any issue related to this problem, please let me know.

closed 2017-03-16 by nemesisdesign

moonwolf-github, 2017-03-31:
Unfortunately i can't compile it:
```
$ make -j1 V=s package/openwisp-config/install
make[1]: Entering directory '/home/moonwolf/Dokumenty/projects/git/lede'
make[1]: *** No rule to make target 'package/openwisp-config/install'.  Stop.
make[1]: Leaving directory '/home/moonwolf/Dokumenty/projects/git/lede'
/home/moonwolf/Dokumenty/projects/git/lede/include/toplevel.mk:199: recipe for target 'package/openwisp-config/install' failed
make: *** [package/openwisp-config/install] Error 2
```
How can i debug this? I was following steps in README.

nemesisdesign, 2017-03-31:
@moonwolf-github it doesn't seem a problem related to this issue.

Try with a [custom firmware image](https://github.com/openwisp/openwisp-config#compiling-a-custom-lede--openwrt-image).

If you keep having problems open a new thread on the [forum/mailing list](https://groups.google.com/d/forum/openwisp/join).