openwisp/openwisp-config #38: Automatic naming of specific sections

nemesisdesign, 2017-06-13:
In order to increase compatibility with the incoming backward conversion feature that will be soon incldued in **OpenWISP 2** (see https://github.com/openwisp/netjsonconfig/issues/70), we need to improve the [openwisp-uci-autoname](https://github.com/openwisp/openwisp-config/blob/5cdb8fa2fc3b7ae9e279c86ac4a82328c5f7c81d/openwisp-config/files/sbin/openwisp-uci-autoname.lua) script so that it can perform the following:

- [ ] ensure `system` config types in `/etc/config/system` are always named `system`
- [ ] ensure `globals` config types in `/etc/config/network` are always named `globals`
- [ ] rename anonymous switches according to the [netjsonconfig switch rendering output](https://github.com/openwisp/netjsonconfig/blob/f27a864e1a038a803b5168225958537b27082642/tests/openwrt/test_network.py#L771-L797)

nemesisdesign, 2017-10-14:
[image] [[openwisp-config] https://github.com/openwisp/openwisp-config/issues/38](https://trello.com/c/Nk9ttaEn/5-openwisp-config-https-githubcom-openwisp-openwisp-config-issues-38)

okraits, 2018-09-12:
Another requirement, especially for `/etc/config/firewall`, is that the order of the sections in the files must be preserved. Right now the order is mixed up which breaks the configuration.

okraits, 2018-09-12:
I will start to work on this tomorrow.

okraits, 2018-09-14:
Todos:

* [x]  ensure `system` config types in `/etc/config/system` are always named `system`
* [x]  ensure `globals` config types in `/etc/config/network` are always named `globals`
* [x]  ensure `defaults` config types in `/etc/config/firewall` are always named `defaults`
* [ ]  rename anonymous switches according to the [netjsonconfig switch rendering output](https://github.com/openwisp/netjsonconfig/blob/f27a864e1a038a803b5168225958537b27082642/tests/openwrt/test_network.py#L771-L797)
* [ ]  ensure anonymous wifi interfaces are named `wifi_<ifname>`
* [ ]  ensure routes are named progressively, eg: `route1`, `route2`, `route3`
* [ ]  ensure rules are named progressively, eg: `rule1`, `rule2`, `rule3`
* [ ]  ensure leds are named as `led_<lowercase_name>`
* [ ]  any other configuration item should be named as `<type><number>`, for example in case of `/etc/config/firewall`, the first forwarding rule will be named: `forwarding1`

nemesisdesign, 2018-09-14:
@okraits I see good progress in the PR!

okraits, 2018-09-18:
I'm done - please review #57.