openwisp/openwisp-config #41: [logging] Add possibility to log to file

nemesisdesign, 2017-11-06:
Until now we have been using the `logger` function available in OpenWRT/LEDE, but we now face the need of logging also to a file.

We need to add a wrapper function which calls logger and also appends to a log file.

okraits, 2019-12-08:
Why do we want to implement logging to a file ourselves? Is it because we want only `openwisp-config` to log to a file and not other services?

I didn't try it out but https://openwrt.org/docs/guide-user/base-system/log.essentials#local_file_logging says that `logd` already provides logging to a file.

nemesisdesign, 2019-12-19:
@okraits I have thought about this recently as well, I'm closing this because it's not essential right now. Thanks for bringing it up.

closed 2019-12-19 by nemesisdesign