openwisp/openwisp-config #48: [feature] new Hook post-reload-hook.

keulu, 2018-04-12:
hello guys

my problem is quite simle, i want to do some stuf when everything is ok and after a new configuration is applied and services reloaded. 

Send message to slack, check openvpn, check coova (and/or restart if necessary, and maybe do other things)

i think with a post reload hook script, we can adjust what the agent can't do.
`
pre_reload_hook
# reload changes and wait $sleep_time
/usr/sbin/openwisp-reload-config
**post_reload_hook**
sleep $sleep_time
`

i can send you a pull request if you think it is a good idea.

nemesisdesign, 2018-04-13:
Hi @keulu,

Thanks for bringing this up. I think it's a good idea, the implementation should be similar to the `pre_reload_hook`.

keulu, 2018-04-16:
ok, i will send you a PR in a couple of days :)

nemesisdesign, 2018-07-18:
Implemented in #52

closed 2018-07-18 by nemesisdesign