openwisp/openwisp-config #58: [idea] Compare release info and update if necessary

nemesisdesign, 2018-09-13:
When upgrading the firmware of a device, the distribution/release info is not updated in OpenWISP.

When the daemon starts, it could compare the information the server has (but there's no API to do this yet, it will need to be added) with the location information, if they don't match it should update the information again (also this will need an API).

nemesisdesign, 2018-09-24:
The idea could be simplified: at startup the deamon could simply send its information to the server.
We need to add an API endpoint to collect this data in [django-netjsonconfig](https://github.com/openwisp/django-netjsonconfig).

okraits, 2020-03-06:
The HW and firmware info (possibly mac address, name or hardware_id, too) also needs to be updated/set when a device is registered which already exists on the server.

nemesisdesign, 2020-03-06:
> The HW and firmware info (possibly mac address, name or hardware_id, too) also needs to be updated/set when a device is registered which already exists on the server.

:+1:

okraits, 2020-03-08:
I will implement that tomorrow as I need it immediately :grinning:

nemesisdesign, 2020-03-09:
@okraits great. I would also make sure the registration URL overwrites the fields if the device registers again, right now it doesn't.

PR #103 (2020-03-25): [agent] Update general device info on the controller after bootup #58
Fixes #58
commit 8c68a52c (2020-03-25): [agent] Update general device info on the controller after bootup #58

Fixes #58
closed 2020-03-25 by nemesisdesign