openwisp/openwisp-config #60: openwisp-uci-autoname.lua: preserve order of options in each section when deleting and re-creating sections

okraits, 2018-09-18:
The script `openwisp-uci-autoname.lua` eliminates anonymous sections by deleting and replacing them with named sections. The order of options in each section is not preserved. Thus if you look at a config file before and after `openwisp-uci-autoname.lua` has run , all the options are jumbled up, which is not nice.

The problem in the code is: when iterating over the items in a table, the order is arbitrary:

https://github.com/openwisp/openwisp-config/blob/master/openwisp-config/files/lib/openwisp/utils.lua#L78

Maybe there is a way to fix this.

nemesisdesign, 2018-09-20:
I just checked, that data structure is returned internally by the lua binding of UCI, I think we'll have to live with that unless there's a more recent way to get the UCI conf as an ordered lua table, so I'm pretty sure the only sane option now is to reorder the options alphabetically..

okraits, 2018-09-21:
Yes, I think so, too. Sorting the options alphabetically is at least deterministic.

commit feede122 (2018-09-21): [lua] write_uci_options in alphabetical order #60

Closes #60
closed 2018-09-21 by nemesisdesign