openwisp/openwisp-config #64: [unmanaged] Handling of unmanaged sections modifies the order of sections

okraits, 2018-09-24:
The handling of unmanaged sections (store/restore) currently doesn't take the order of sections into account. In the firewall config this can lead to a modified order of rules and thus to a different behaviour.

nemesisdesign, 2018-09-24:
#65 is merged, now we should remember to change the default `unmanaged` value so this feature is disabeld because I believe almost nobody is using it.

nemesisdesign, 2018-09-25:
Now that #67 is done, I was thinking to close this but maybe we leave it open because the issue is not really resolved? We just worked around it by disabling it since it's probably not used very much. But if someone uses it on configuration that need to maintain their order they'll incur in this issue.

okraits, 2018-09-25:
:+1: 
Yes, this still needs to be fixed. Maybe the handling of unmanaged sections needs to be changed in general - I don't see any easy solution to prevent reordering of sections with the way it's done right now.

nemesisdesign, 2018-09-25:
@okraits I think the index can be stored and sections can be re-inserted in the same place, but there's no advantage for us in doing this now since we don't use this feature

okraits, 2018-09-26:
Yes, indexes need to be stored in any case. I guess there could be scenarios where the configuration update modifies the order of sections (or creates new sections) so that a collision happens between the indexes of the unmanaged sections and the indexes of the rest of the configuration.

But we need to look into that anyway when we tackle this issue.

okraits, 2019-05-09:
I think you can only declare whole sections and specific section types as unmanaged section, not single options.
What are those `id` parameters for? What do you want to achieve with them?