openwisp/openwisp-config #72: [update-config] Create a backup of the local UCI configuration to be able to set it back to its original state

okraits, 2018-10-05:
Right now openwisp-config doesn't keep a backup of the local UCI configuration (all files in `/etc/config`). 

This can lead to situations where settings are lost:

1. An existing section is modified by openwisp-config.
2. The modifications are removed from the configuration tarball delivered by openwisp-controller.
3. openwisp-config deletes this section instead of restoring the state it had before it was modified by OW2.

Thus a backup of the initial configuration should be created and restored when the modifications are removed.

nemesisdesign, 2018-10-05:
Once this, #71 and #69 are fixed the config update process will be improved greatly!

BlancLoup, 2018-11-25:
**Problem:**
IMHO that it's not the right way. The real problem is backup realized like "sysupgrade -b". Sysupgrade does not match "principles" of apply_configuration function. Apply_configuration does correct actions for "diff-like" configurations while sysupgrade saves the entire configuration (including /etc/config/openwisp).

**Consequences:** 
1. apply_configuration always restores /etc/config/openwisp that leads to openwisp agent restarting before restore process was completed. Integrity will be ruined.
2. all daemons will be restarted even that is not need.

**Solution (backup only files that will be changed, very simpled algorithm):**
1. check new configuration archive for files that will be changed:
**_tar -ztf $configuration_archive_to_apply | sed 's/^/\//' > $list_of_files_will_be_changed_**
2. create backup archive of files that will be changed
**_tar -czvf $backup_of_current_files -T $list_of_files_will_be_changed_**

okraits, 2018-11-28:
I don't understand what you're trying to say. The use of sysupgrade hasn't been changed and it's not involved here at all.

The issue addressed here is that if openwisp-config modifies an element twice then it has no history available to tell wether the element was originally either added or modified by openwisp-config. Thus it is not able to restore the original state. This change fixes that.

BlancLoup, 2018-11-28:
Sorry... I mixed up a little bit.

okraits, 2018-11-29:
If you have valid and serious concerns about my changes then please speak up and elaborate. It might be possible that I failed to get your point. Maybe @nemesisdesign can also add his take on this.

nemesisdesign, 2018-12-02:
@BlancLoup where you mixing this issue with #81?

Gobiviswa, 2019-06-05:
Hi, I am just thinking what is the need of deleting the files when update the configurations? is there any specific reason for that? By the way, is this issue resolved ? 
Thanks,

okraits, 2020-02-07:
Today a workmate of mine had the issue that an outdated backup of `/etc/config/openwisp` was present in `/etc/openwisp/stored/etc/config/openwisp`.

When he applied changes to `/etc/config/openwisp` and later removed those changes, the outdated backup was used to restore the original state.

Which led to the issue that `openwisp-config` used a wrong uuid/key to fetch the checksum from the controller. I have to look into that and fix it.

commit 31c530c4 (2021-09-14): [update-config] Store and restore original UCI configuration #72

Fixes #72
commit faca3cc7 (2021-10-04): [update-config] Store and restore original UCI configuration #72

Fixes #72
commit c7359130 (2021-10-12): [update-config] Store and restore original UCI configuration #72

Fixes #72