openwisp/openwisp-config #74: [runbuild] Differentiate between nightly builds and releases

nemesisdesign, 2018-10-09:
We should make the [runbuild script](https://github.com/openwisp/openwisp-config/blob/master/runbuild) able to differentiate between releases (tagged commits) and nightly builds, which would have to be published to different directories.

Subsequent steps will need to update the README, the docs on openwisp.io and release more often.

devkapilbansal, 2022-05-26:
Hi @nemesisdesign , is this issue still relevant?

If yes, why we need a different directory for tagged ones?
Also, what are the changes that it will make in README or docs?

nemesisdesign, 2022-05-27:
@devkapilbansal I am not so sure right now, I think there are other issues we can work on which would bring us more benefit. Right now the main thing we are missing on the downloads.openwisp.io is the checksum for each compiled package, there's no way for the users to be 100% to verify the package they downloaded hasn't been tampered with.