openwisp/openwisp-config #77: [openwisp-config] Add luacheck to runtests script and fix all warnings

nemesisdesign, 2018-10-21:
**Reserved for GCI**

Luacheck is a static analyzer and a linter for Lua.

It is possible to integrate luacheck in many popular editors, like atom, vim, VS Code.

Prepend the luacheck test in the openwisp-config `runtests` script (add right after `cd openwisp-config/tests/`) and fix all warnings and errors luacheck reports.

nemesisdesign, 2018-10-21:
@okraits I merged the two tasks into one becuase it would be time-consuming to enforce having two tasks in which one is dependent on the other (we may get two students claiming both tasks concurrently) so in this case it made sense to merge the two in one

okraits, 2018-12-06:
@nemesisdesign I had a look at this and noticed some things:

* Only the tests are checked by `luacheck`, not the actual module code
* Warnings about redefined variables, implicitly defined variables and unused arguments and loop variables are suppressed - but those can still lead to bugs and unused code
* No warnings and errors have been fixed in the actual module code
* thus luacheck integration in editors still reports many warnings and errors as those warnings mentioned above are not suppressed normally

I'm not sure if this is wanted. But maybe I'm wrong.

okraits, 2018-12-07:
So let's reopen this issue so that things can get improved.

nemesisdesign, 2019-11-27:
@okraits is this still relevant?

okraits, 2019-11-27:
@nemesisdesign Yes, of course.

TPath123, 2019-12-05:
Was planning on working on this, just had a few queries:
> * Only the tests are checked by `luacheck`, not the actual module code
Is this something that needs to be fixed (I couldn't tell if this was wanted or not)
Also, are there any lints that you still wish to skip (currently -d -a -r flags are set)?

okraits, 2019-12-05:
> > * Only the tests are checked by `luacheck`, not the actual module code
> Is this something that needs to be fixed (I couldn't tell if this was wanted or not)

This is not wanted and needs to be fixed.

> Also, are there any lints that you still wish to skip (currently -d -a -r flags are set)?

I would remove all three flags if possible.

okraits, 2019-12-05:
For convenience, you can keep the `-a` flag for now.

TPath123, 2019-12-06:
@okraits 
I've come up with the following notes whilst working on this issue:

- Old Modules
The modules "uci" and "lfs" etc. are based on the old Lua 5.0/5.1 way of creating modules (`modules("*", package.seeall)`) - Is this something that should be updated?
This also triggers luacheck errors as these modules rely on importing the functions into a global namespace

- Tests require globals definitions
The unit tests require the testing table be a global variable, however this triggers a fail with Luacheck (See below for resolution)

**Idea:**
I was thinking, how about introducing a `.luacheckrc` file in regards to fixing a few of these problems, and also allowing us to remove the `-a` flag?

The file could be set out as such:
```lua
ignore = {"213/_.*"} -- unused Loop Variables beginning with underscore will be ignored, keeping readability
files["openwisp-config/tests"] = {
    ignore = {"11./Test.*"} -- globals relating to defining unittests are ignored
}
```
This will ensure a strict code linting, whilst also working around the issues we currently have.

okraits, 2019-12-07:
> * Old Modules
>       The modules "uci" and "lfs" etc. are based on the old Lua 5.0/5.1 way of creating modules (`modules("*", package.seeall)`) - Is this something that should be updated?
>       This [...]

This means that those reported luacheck errors are in the external modules we use, not in our code, right? That would mean that those errors have to be fixed upstream. This has to be done, but it is out of scope of this PR:

> * Tests require globals definitions
>       The unit tests require the testing table be a global variable, however this triggers a fail with Luacheck (See below for resolution)
>
> **Idea:**
> I was [...]

A very good idea! Go ahead! :grinning:

PR #80 (2018-11-11): Added Luacheck and travis.yml with install-config 
### Files added/changed - _Summary_ 
- added travis.yml with **dependencies installation configuration.**
- added luacheck test command with parameters (_-d -a -r_) to filter unwanted warnings in **runtests script**

- after running runtests script
 -- fixed **`accessing undefined variable uci`** warning by making **`uci`** global variable and assigning to `require('uci')`in `openwisp-config/tests/test_utils.lua` file

fixes #77 

this is a submission for **GCI task**- **Add travis-ci configuration for luacheck - OpenWISP**
commit 83a377d5 (2018-11-11): [travis.yml] added install-script & luacheck install command in travis. #77

- added travis.yml with dependencies installation script.
- added luacheck test command in runtests script.
- fixed accessing undefined variable `uci` warning

Fixes #77
PR #96 (2019-12-11): [luacheck] Run luacheck on all files and fix warnings
Removes all filter flags from luacheck, and ensure it runs over the whole codebase
Fix all luacheck warnings that arise from this.
Closes #77
commit bc94e24c (2019-12-11): [ci] Add luacheck to tests #77

Remove all filter flags from luacheck
Ensure luacheck tests over all code

Fixes #77
closed 2019-12-11 by nemesisdesign