openwisp/openwisp-config #86: [Makefile] Update makefile in the OpenWrt packages feed

okraits, 2019-01-18:
The makefile in the OpenWrt packages feed has to be updated:

https://github.com/openwrt/packages/tree/master/admin/openwisp-config

Things to consider:

* Do we want to have two separate makefiles? (One here and one in the packages feed) I guess we want to be in the OpenWrt packages feed for distribution and to increase our visibility in the OpenWrt ecosystem.
* Can the two makefiles be identical or are there things which have to be adapted on each update?

nemesisdesign, 2019-01-18:
I think that if we can make them identical, would be great.
If we can find a way to have only the `Makefile` in the openwrt repo and pull everything from here in some way, even better. I'm not an expert with that.

BKPepe, 2020-11-06:
I think having just one Makefile in the OpenWrt packages feed would be enough. Right now, it increases your maintenance efforts by having two Makefiles and it will avoid issues like this https://github.com/openwisp/openwisp-config/issues/120. Because this issue was not present in the packages feed.

nemesisdesign, 2020-11-06:
> I think having just one Makefile in the OpenWrt packages feed would be enough. Right now, it increases your maintenance efforts by having two Makefiles and it will avoid issues like this #120. Because this issue was not present in the packages feed.

@BKPepe I'm not sure how that would work in practice, do you have a few examples?
For example, how do we compile new versions of this package when we make changes then?

I'm also worried about the fact that OpenWISP has been evolving rapidly lately, so having to wait for others to merge changes may not work for us in this phase.

nemesisdesign, 2020-12-20:
Opened a PR https://github.com/openwrt/packages/pull/14283.
@okraits could you help me to make sure all the checks there pass and test it?

nemesisdesign, 2021-01-18:
https://github.com/openwrt/packages/pull/14283 has been merged!
Now waiting for https://github.com/openwrt/packages/pull/14499, which will add it to openwrt-19.07.

nemesisdesign, 2021-01-19:
Also https://github.com/openwrt/packages/pull/14499 was merged.

closed 2021-01-19 by nemesisdesign