openwisp/openwisp-config #87: Failed to connect to controller during registration: curl exit code 6

JohnnyKetch, 2019-04-01:
hi
I'm trying to register my OpenWRT device but I still get these two errors:

`Mon Apr  1 02:53:51 2019 daemon.info openwisp: Registering device...`
`Mon Apr  1 02:53:51 2019 daemon.err openwisp: Failed to connect to controller during registration: curl exit code 6`

Questo è il mio file di configurazione su OpenWRT:

`config controller 'http'`
`        option url 'https://diocane/'  **//local server** `
`        #option interval '120'`
`        option verify_ssl '0'`
`        option shared_secret 'gSrXrkd1YR9StAfqWgBLv1wbPjzXEnDN'`
`        #option consistent_key '1'`
`        #option mac_interface 'eth0'`
`        #option merge_config '1'`
`        #option test_config '1'`
`        #option test_script '/usr/sbin/mytest'`
`        #option uuid ''`
`        #option key ''`
`        list unmanaged 'system.@led'`
`        list unmanaged 'network.loopback'`
`        list unmanaged 'network.@switch'`
`        list unmanaged 'network.@switch_vlan'`
`        # curl options`
`        #option connect_timeout '15'`
`        #option max_time '30'`
`        #option capath '/etc/ssl/certs'`
`        # hooks`
`        #option pre_reload_hook '/usr/sbin/my_pre_reload_hook'`

I have tried both with the "0.4.5-2" version and the last one :(
`

okraits, 2019-04-01:
Exit code 6 means: `Couldn't resolve host. The given remote host was not resolved.`
Your url is not valid: `'https://diocane/' **//local server**'`. In most cases this should be the url you use to access the OpenWISP webinterface.

nemesisdesign, 2019-04-01:
This is not a bug but a configuration issue, please use the [support channels](http://openwisp.org/support.html) to get help; we use this to track bugs.

The mailing list is still the best place to ask questions for the reasons described here: http://openwisp.io/docs/general/help-us.html#write-to-the-mailing-list

closed 2019-04-01 by nemesisdesign