openwisp/openwisp-config #89: [apply-config] order of items in a list is not set as defined in NetJSON

okraits, 2019-07-18:


nemesisdesign, 2019-07-18:
How to replicate the issue? Is the configuration on the controller changed?

okraits, 2019-07-19:
1. Make sure there are no ntp servers defined.
2. Add 3 ntp servers in the device config or in a template.
3. When the config is applied on the device, the order of ntp servers is not set as defined on the controller.

katsmark, 2019-07-25:
Hello.
I am a complete newbie in this field. I have a question. Why are we not using the entire list sent from the controller, but adding items from it to the existing list? What if we need to remove items from the existing list? For example, we want to remove one of the ntp servers from the default list.
Thanks!

nemesisdesign, 2019-12-19:
@katsmark: https://github.com/openwisp/openwisp-config/issues/82

commit 78a40c78 (2020-09-21): [fix/change!] List options are now overwritten #82 #89

When using list options the old behavior was to merge
the options present in the controller with the ones
present locally, which made removing list options not
possible without hacky workarounds.

This change fixes the problem by ensuring list options
present in the controller overwrite the local ones.

Closes #82

It also solves issue https://github.com/openwisp/openwisp-config/issues/89
because list options do not go through the process of being converted
to a set and then back to a list (which shuffles the order).

Closes #89
PR #115 (2020-09-21): [fix] Fix removal of list options #82
When using list options the old behavior was to merge
the options present in the controller with the ones
present locally, which made removing list options not
possible without hacky workarounds.

This change fixes the problem by ensuring list options
present in the controller overwrite the local ones.

Closes #82

It also solves #89 because list options do not go through the process
of being converted to a set and then back to a list (which shuffles the order).
Closes #89
closed 2020-09-21 by nemesisdesign