openwisp/openwisp-config #105: [setup] runtests failed on setup

atb00ker, 2020-04-10:
While installing openwisp-config; I did:

    sudo ./install-dev.sh
    ./runtests

and I am getting 38 warnings, is there any step I am missing?

Logs: 
```
atb00ker@pavilion:~/Documents/develop/openwisp/openwisp-config$ ./runtests
Checking json-c/libubox/examples/uloop-example.lua 15 warnings

    json-c/libubox/examples/uloop-example.lua:14:10: setting non-standard global variable t
    json-c/libubox/examples/uloop-example.lua:18:21: accessing undefined variable t
    json-c/libubox/examples/uloop-example.lua:28:10: setting non-standard global variable p1
    json-c/libubox/examples/uloop-example.lua:33:10: setting non-standard global variable p2
    json-c/libubox/examples/uloop-example.lua:40:69: accessing undefined variable p1
    json-c/libubox/examples/uloop-example.lua:45:69: accessing undefined variable p2
    json-c/libubox/examples/uloop-example.lua:49:1: setting non-standard global variable udp_ev
    json-c/libubox/examples/uloop-example.lua:53:3: accessing undefined variable udp_ev
    json-c/libubox/examples/uloop-example.lua:57:1: setting non-standard global variable udp_count
    json-c/libubox/examples/uloop-example.lua:58:1: setting non-standard global variable udp_send_timer
    json-c/libubox/examples/uloop-example.lua:62:6: accessing undefined variable udp_count
    json-c/libubox/examples/uloop-example.lua:64:4: accessing undefined variable udp_send_timer
    json-c/libubox/examples/uloop-example.lua:67:4: accessing undefined variable udp_send_timer
    json-c/libubox/examples/uloop-example.lua:73:3: setting non-standard global variable udp_count
    json-c/libubox/examples/uloop-example.lua:73:15: accessing undefined variable udp_count

Checking json-c/libubox/uci/tests/cram/lua/basic.lua 2 warnings

    json-c/libubox/uci/tests/cram/lua/basic.lua:2:11: accessing undefined variable uci
    json-c/libubox/uci/tests/cram/lua/basic.lua:41:7: variable t was previously defined on line 11

Checking json-c/libubox/uci/tests/cram/lua/test_cases/changes_doesnt_leak.lua 1 warning

    json-c/libubox/uci/tests/cram/lua/test_cases/changes_doesnt_leak.lua:2:11: accessing undefined variable uci

Checking json-c/libubox/uci/tests/cram/lua/test_cases/set_with_empty_table_doesnt_leak.lua 1 warning

    json-c/libubox/uci/tests/cram/lua/test_cases/set_with_empty_table_doesnt_leak.lua:1:11: accessing undefined variable uci

Checking libubox/examples/uloop-example.lua       15 warnings

    libubox/examples/uloop-example.lua:14:10: setting non-standard global variable t
    libubox/examples/uloop-example.lua:18:21: accessing undefined variable t
    libubox/examples/uloop-example.lua:28:10: setting non-standard global variable p1
    libubox/examples/uloop-example.lua:33:10: setting non-standard global variable p2
    libubox/examples/uloop-example.lua:40:69: accessing undefined variable p1
    libubox/examples/uloop-example.lua:45:69: accessing undefined variable p2
    libubox/examples/uloop-example.lua:49:1: setting non-standard global variable udp_ev
    libubox/examples/uloop-example.lua:53:3: accessing undefined variable udp_ev
    libubox/examples/uloop-example.lua:57:1: setting non-standard global variable udp_count
    libubox/examples/uloop-example.lua:58:1: setting non-standard global variable udp_send_timer
    libubox/examples/uloop-example.lua:62:6: accessing undefined variable udp_count
    libubox/examples/uloop-example.lua:64:4: accessing undefined variable udp_send_timer
    libubox/examples/uloop-example.lua:67:4: accessing undefined variable udp_send_timer
    libubox/examples/uloop-example.lua:73:3: setting non-standard global variable udp_count
    libubox/examples/uloop-example.lua:73:15: accessing undefined variable udp_count

Checking libubox/uci/tests/cram/lua/basic.lua     2 warnings

    libubox/uci/tests/cram/lua/basic.lua:2:11: accessing undefined variable uci
    libubox/uci/tests/cram/lua/basic.lua:41:7: variable t was previously defined on line 11

Checking libubox/uci/tests/cram/lua/test_cases/changes_doesnt_leak.lua 1 warning

    libubox/uci/tests/cram/lua/test_cases/changes_doesnt_leak.lua:2:11: accessing undefined variable uci

Checking libubox/uci/tests/cram/lua/test_cases/set_with_empty_table_doesnt_leak.lua 1 warning

    libubox/uci/tests/cram/lua/test_cases/set_with_empty_table_doesnt_leak.lua:1:11: accessing undefined variable uci

Checking openwisp-config/files/lib/openwisp/net.lua OK
Checking openwisp-config/files/lib/openwisp/utils.lua OK
Checking openwisp-config/files/sbin/openwisp-get-address.lua OK
Checking openwisp-config/files/sbin/openwisp-remove-default-wifi.lua OK
Checking openwisp-config/files/sbin/openwisp-restore-unmanaged.lua OK
Checking openwisp-config/files/sbin/openwisp-store-unmanaged.lua OK
Checking openwisp-config/files/sbin/openwisp-uci-autoname.lua OK
Checking openwisp-config/files/sbin/openwisp-update-config.lua OK
Checking openwisp-config/tests/test_remove_default_wifi.lua OK
Checking openwisp-config/tests/test_restore_unmanaged.lua OK
Checking openwisp-config/tests/test_store_unmanaged.lua OK
Checking openwisp-config/tests/test_uci_autoname.lua OK
Checking openwisp-config/tests/test_update_bad_config.lua OK
Checking openwisp-config/tests/test_update_config.lua OK
Checking openwisp-config/tests/test_utils.lua     OK

Total: 38 warnings / 0 errors in 23 files
atb00ker@pavilion:~/Documents/develop/openwisp/openwisp-config$
```

nemesisdesign, 2020-04-10:
Looking at it again, first impression is that the lua linter is raising warnings for lua files that are dependency of openwisp-config, eg: `json-c/libubox/examples/uloop-example.lua`

atb00ker, 2020-04-11:
Yes, after inspection, I changed `luacheck ./ -a` -> `luacheck ./openwisp-config/ -a` to ignore the "dev-install" files and everything is working as expected. :smile:

atb00ker, 2020-04-11:
Realized the problem was in install script. Made the change in the PR: https://github.com/openwisp/openwisp-config/pull/106#issue-402187566

PR #106 (2020-04-11): [setup] fixed install & runtests scripts
Fixed setup problems.
1. In `install-dev.sh` installation procedure of json-c needed to be updated. :-)

2. Failing chained commands were not stopping the install script immediately, changed that by using `command || {exit 1}` trick.

3. In `./runtests` scripts, changed `luacheck . -a` -> `luacheck ./openwisp-config/ -a` because if I do `./runbuild` with BUILD_DIR as `~/.build`, runtests wouldn't work again, I suppose other developers would want to build for testing in this directory itself as well, hence made this change.
Let me know if BUILD_DIR shoul [... cut]
commit 677affdb (2020-04-11): [setup] fixed install & runtests scripts #105

Closes #105
closed 2020-04-11 by nemesisdesign