openwisp/openwisp-config #136: Restarting a device messes up with firewall configuration

apoulias, 2021-05-26:
I have a device with the default OpenWrt firewall configuration (lan + wan) running openwisp-config. 

Adding a new firewall zone from Luci and restarting the device, triggers the openwisp-uci-autorename which returns the following message:
"The following uci configs have been renamed: firewall.zone1".

The problem is that the zone1 is the lan zone which is not anonymous and it's messed up with the new one.

The initial config/firewall contains the following:
config zone 'zone1'
        option forward 'ACCEPT'
        option input 'ACCEPT'
        option name 'lan'
        list network 'lan'
        option output 'ACCEPT'

config zone 'zone2'
        option forward 'REJECT'
        option input 'REJECT'
        option masq '1'
        option mtu_fix '1'
        option name 'wan'
        list network 'wan'
        list network 'wan6'
        option output 'ACCEPT'

config zone               
        option input 'ACCEPT'
        option forward 'REJECT'
        option name 'newZone'
        option output 'ACCEPT'

After restarting:
config zone 'zone2'
        option forward 'REJECT'
        option input 'REJECT'
        option masq '1'
        option mtu_fix '1'
        option name 'wan'
        list network 'wan'
        list network 'wan6'
        option output 'ACCEPT'

config zone 'zone1'            
        option input 'ACCEPT'
        list network 'lan'
        option output 'ACCEPT'
        option forward 'REJECT'
        option name 'newZone'

I've figured it out that if the new zone isn't anonymous (config zone 'newZone') the problem doesn't appear, which led me to openwisp-uci-autorename script.

apoulias, 2021-05-28:
In case this is actually a bug, I'd like to work on that.

nemesisdesign, 2021-06-03:
Yes it's most probably `openwisp-uci-autorename`, let me know what you can find.

apoulias, 2021-06-03:
The problem lies with the way the new name of an unnamed config is obtained. The incCount doesn't take care of the already existing configurations like the example in the issue description. When the script identified the first unnamed config (newZone) the counter became 1 and the assigned name 'zone1' which already existed. 
To reproduce, assign a name to the first route config in file openwisp-config/tests/config/network as 'route1' and run the tests.
I've implemented another way in my fork which always makes sure that the new name would not be duplicate. I'll create a pull request for a review.

commit 75f352c0 (2022-05-09): [fix] Fixed conflict in renaming of anon UCI confs #136

The algorithm for renaming anonymous configurations may use the
name of an existing one, if that was named and some conditions were met.

For example:

config route 'route1'
        option test '1'
config route
        option test '2'

would result in:

config route 'route1'
        option test '2'

This patches fixes this issue.

Closes #136.
closed 2022-05-09 by nemesisdesign