openwisp/openwisp-config #167: [bug] Bootup delay currently broken on master

nemesisdesign, 2022-03-26:
See my comment here: https://github.com/openwisp/openwisp-config/commit/b150e658541f4e58e918309e437b5ddbb104459f#r69603100

I tried also using the syntax of the previous version but it's not working.

@okraits @devkapilbansal FYI.

Found out thanks to this report #166.

okraits, 2022-03-26:
This looks like a regression. Has it been tested after the changes have been made?

I haven't looked deeper into this but `-ne` looks like it should be applied to an integer. But `"0"` isn't an integer, it's a string.

devkapilbansal, 2022-03-27:
> I haven't looked deeper into this but `-ne` looks like it should be applied to an integer. But `"0"` isn't an integer, it's a string.

I don't think this is the reason. `"0"` is also treated as an integer :thinking:

nemesisdesign, 2022-03-27:
> This looks like a regression. Has it been tested after the changes have been made?

It seems it wasn't tested.
But I did test it on Friday.

> I haven't looked deeper into this but `-ne` looks like it should be applied to an integer. But `"0"` isn't an integer, it's a string.

This is not the issue.
The code block gets executed, but the value of `DELAY` is always zero I tried this on OpenWrt 21, I tried also the old code and it also does not work, it could have to do something with my device or with OpenWrt 21 / or better said the busybox version used by OpenWrt which may have changed this part of the random generation of numbers.

momothefox, 2022-03-28:
@nemesisdesign 

> it could have to do something with my device or with OpenWrt 21 

Tested on Openwrt 19.07.9 same issue. 
OpenWrt 19.07.9 r11405-2a3558b0de / LuCI openwrt-19.07 branch git-22.045.73925-36e5c1c
Openwisp version 0.5.0
Device TP-Link Archer C50 v4

okraits, 2022-03-28:
I just tested the bootup delay in openwisp-config based on https://github.com/openwisp/openwisp-config/commit/8b85ca5eadb09686c6d6f010e8a3b716061b507e on OpenWrt 19.07 and it works fine.

So the issue must have appeared through changes after that.

momothefox, 2022-03-28:
@okraits 
lookup this code 
[https://serverfault.com/questions/347582/adding-a-random-delay-for-a-linux-command](url)
i had to try and error untill i got to this working 19.07
if [ ! -f "$BOOTUP" ]; then
	# actions to do only after agent start, not after registration or agent restart
	if [ "$BOOTUP_DELAY" -ne "0" ]; then
		# random bootup delay between 0..BOOTUP_DELAY seconds
		DELAY="$((BOOTUP_DELAY+RANDOM % BOOTUP_DELAY))"
		logger "Delaying startup for maximum $DELAY seconds..." \
				-t openwisp \
				-p daemon.info
		sleep $DELAY
	fi
	# send up to date device info
	update_info
	touch "$BOOTUP"
fi

okraits, 2022-03-28:
I don't know what I'm missing, but it's working fine here also with a quite recent version: https://github.com/openwisp/openwisp-config/commit/d65be8c26d5e6cabf31fb8f37f8aba089622e452 on OpenWrt 21.02.

The RANDOM function is also working as expected if called from a shell:

```
root@VR-2020 /etc/config # BOOTUP_DELAY=60
root@VR-2020 /etc/config # echo "$((RANDOM % BOOTUP_DELAY))"
35
root@VR-2020 /etc/config # echo "$((RANDOM % BOOTUP_DELAY))"
51
root@VR-2020 /etc/config # echo "$((RANDOM % BOOTUP_DELAY))"
10
root@VR-2020 /etc/config # echo "$((RANDOM % BOOTUP_DELAY))"
50
root@VR-2020 /etc/config # echo "$((RANDOM % BOOTUP_DELAY))"
34
root@VR-2020 /etc/config # echo "$((RANDOM % BOOTUP_DELAY))"
59
```

okraits, 2022-03-28:
We enable Ash random support by setting `BUSYBOX_DEFAULT_ASH_RANDOM_SUPPORT=y`. In upstream OpenWrt this is not enabled.
So we will need to change the implementation to get along without RANDOM.

momothefox, 2022-03-28:
@okraits https://stackoverflow.com/questions/4678836/how-to-generate-random-numbers-under-openwrt

nemesisdesign, 2022-03-28:
> We enable Ash random support by setting `BUSYBOX_DEFAULT_ASH_RANDOM_SUPPORT=y`. In upstream OpenWrt this is not enabled: https://github.com/openwrt/openwrt/blob/master/package/utils/busybox/Config-defaults.in#L2994 So we will need to change the implementation to get along without RANDOM.

This explains it, thanks. 

We could use awk, there's a few examples on stackoverflow, will the seed be good enough?
Alternatively we can add a lua script.

okraits, 2022-03-30:
You can't do float math in ash and `bc` is not available either. I suggest using a lua script:

* Call math.randomseed with some input from _/dev/urandom_
* Use math.random to get a random number

devkapilbansal, 2022-04-22:
> @okraits https://stackoverflow.com/questions/4678836/how-to-generate-random-numbers-under-openwrt

@nemesisdesign This works fine. We can use this instead of RANDOM

nemesisdesign, 2022-04-27:
A solution is almost ready in https://github.com/openwisp/openwisp-config/pull/171.

commit 7a9279e7 (2022-04-25): [Fix] Generate random number using awk in bootup delay #167

Fixes #167
commit 87cfa664 (2022-04-26): [Fix] Generate random number using lua script in bootup delay #167

Fixes #167
PR #171 (2022-04-28): [Fix] Generate random number using lua script in bootup delay #167
Fixes #167
commit 4109683e (2022-04-28): [fix] Fixed bootup_delay #167

Fixes #167
closed 2022-04-28 by nemesisdesign