openwisp/openwisp-config #181: [feature] Add SIGUSR2 which forces the reload of the config

nemesisdesign, 2022-06-25:
We could listen to SIGUSR2 which would be like SIGUSR1 but with the difference that the following files will be removed first:

- /tmp/openwisp/checksum
- /etc/openwisp/checksum

This will force the reload of the configuration. 

It would be super great to also provide some sort of shortcut to send these signals to the agent, easily, with a command that we can remember, what about:

```
openwisp_config --check-conf
openwisp_config --force-conf
```

I find myself doing these kind of things all the time manually, I am sure many users do the same or struggle to find out how to do this, so if we add an easy way and document it, it will be useful.

shwetd19, 2025-02-11:
Hey @nemesifier,  

Implementation I can think of: Adding a signal handler for `SIGUSR2` in the agent, and when the signal is received, delete the checksum files (`/tmp/openwisp/checksum` and `/etc/openwisp/checksum`). Additionally, also just to confirm here... adding CLI commands like `openwisp_config --check-conf` and `openwisp_config --force-conf` for ease of use, and document the new feature in the user guide can solve this manual work here..

nemesifier, 2025-02-11:
> Hey [@nemesifier](https://github.com/nemesifier),
>
> Implementation I can think of: Adding a signal handler for `SIGUSR2` in the agent, and when the signal is received, delete the checksum files [...]

@shwetd19 sounds good! The agent name now is `openwisp-config` (before it was `openwisp_config`).