openwisp/openwisp-config #186: [feature] Add retry mechanism with randomized backoff for HTTP requests

nemesisdesign, 2022-11-04:
HTTP requests can fail.
We need to add/modify any existing retry mechanism to make sure we have a randomized back off between 2 and 15 seconds for:

- GET checksum (here there's a retry mechanism for 404 which can stay intact, we can add slightly different logic for the normal case)
- GET config
- POST report status

It would be good to implement this as a reusable function which we can use in the different places of the code.

pandafy, 2022-11-25:
This is the current state for flow (ignoring the 404 handling for Get checksum): 

1. The agents calls `configuration_changed` and `update_configuration` from this loop 
https://github.com/openwisp/openwisp-config/blob/5798e714fb4229a757cee5cbc006a7139308a9f2/openwisp-config/files/openwisp.agent#L930-L948
2. If any error is encountered in these operations (fetching checksum, downloading configuration or applying configuration, the agent restores previous state on the device and wait till next cycle in "$INTERVAL" time. 

We can add the retry back-off here in the interval instead of adding it at different places all over the code. This way we can handle `GET checksum` and `GET config` from one place

```diff
diff --git a/openwisp-config/files/openwisp.agent b/openwisp-config/files/openwisp.agent
index 5e18b1c..496fc79 100755
--- a/openwisp-config/files/openwisp.agent
+++ b/openwisp-config/files/openwisp.agent
@@ -932,16 +932,22 @@ while true; do
        # (because the address may change due to
        # configuration updates or manual reconfigurations)
        discover_management_ip
-
-       configuration_changed
+       local config_changed_exit_code update_config_exit_code retry_backoff
+       retry_backoff=0
+       config_changed_exit_code=configuration_changed
 
        if [ "$?" -eq "1" ]; then
-               update_configuration
+               update_config_exit_code=update_configuration
+       fi
+
+       if [[ "$config_changed_exit_code" -ne "1" && "$config_changed_exit_code" -ne "0" ]] \
+               || [ $update_config_exit_code -eq "3" ]; then
+               retry_backoff=$(/usr/sbin/openwisp-get-random-number 2 15)
        fi
 
        # handle SIGUSR1 to interrupt sleep
        trap handle_sigusr1 USR1
-       sleep "$INTERVAL" &
+       sleep $((INTERVAL + retry_backoff)) &
        wait $!
        # ignore SIGUSR1 signals again
        trap "" USR1
```

For `POST report status` we will have to use a more local approach.

okraits, 2022-11-25:
What do you mean by `backoff`?

> HTTP requests can fail. We need to add/modify any existing retry mechanism to make sure we have a randomized delay between each retry, the random delay being between 2 and 15 seconds for the following requests:

So for GET checksum, the random delay will be added to the configured `interval` value, right?

PR #188 (2022-12-20): [feature] Added retry mechanism with random backoff #186
Added retry mechanism with random backoff for following functions:
- get_checksum
- download_configuration
- report_status

Closes #186

-------------

I have manually tested these changes with my device.
closed 2022-12-20 by nemesisdesign