openwisp/openwisp-config #195: [bug] The openwisp-config UCI configuration should not be restored completely

nemesifier, 2023-06-27:
If a device is switched from one instance to another, it may happen that the agent stores a backup of the full /etc/config/openwisp file, which will contain URL, UUID, key and in some cases shared secret.
In some cases this causes the backup to be restored after registering to a different server, which basically causes a rollback and makes the device go back to the old server.

A possible solution could be to treat this file differently and avoid storing URL, UUID, key or shared secret but before proceeding with this we have to verify there's no other downsides to this approach.

KaizokuHam, 2026-10-02:
Hi @nemesifier, I'd like to work on this issue.

I looked into `backup_configuration()` and `restore_backup()` in `openwisp.agent`. From what I understand, restoring a previous backup could also restore outdated OpenWISP credentials, causing a device to reconnect to its previous controller.

I'd like to investigate how `/etc/config/openwisp` gets included in the backup and reproduce the behavior before proposing a fix.

Would preserving the current controller URL and registration credentials during restoration be the right direction, or is there another approach you'd prefer?

Is this issue available to work on?

mchaitanyanair-del, 2026-11-04:
Hi! I've analyzed this issue and would like to work on it.

Downsides Analysis:
Stripping url, uuid, key, and shared_secret from agent backups prevents migration rollbacks, but we must ensure that:

Offline manual restores by technicians don't lose device identity entirely (or verify that re-registration handles missing keys gracefully).

The UCI parser daemon scripts handle missing keys without throwing unhandled exceptions or breaking the loop.

I plan to implement a selective sanitization filter in the backup routine and test it against the agent's re-registration lifecycle. 
Could you please assign this issue to me?

@openwisp-companion assign

mchaitanyanair-del, 2026-11-08:
Hey , currently reviewing the staging logic and boundary conditions for #195 to ensure we have the cleanest POSIX implementation. Will be pushing the verified updates shortly!