openwisp/openwisp-config #223: [bug]  Instance openwisp_config::instance1 s in a crash loop x

kellerman, 2024-10-17:
**Describe the bug**
Hello everyone!
I recently upgraded the package of openwisp-config to version 1.1.0-1 in OpenWrt 23.05.5. I couldn't connect to the controller anymore.
In logs this message appears (and controller doesn't see the client):
`Instance openwisp_config::instance1 s in a crash loop 6`
I downgraded to 1.0.1 using this package from 22.03.7 and everything was back to normal:
- https://downloads.openwrt.org/releases/22.03.7/packages/mips_24kc/packages/openwisp-config_1.0.1-1_all.ipk

**Steps To Reproduce**
Steps to reproduce the behavior:
opkg update
opkg install openwisp-config openwisp-monitoring luci-app-openwisp
#check system log for crash info

**Expected behavior**
There is no crashes and controller recognizes the client.

**System Informatioon:**
 - OpenWrt Version: 23.05.5

diepeterpan, 2024-10-18:
Jip, I fixed my by manually download [https://downloads.openwrt.org/releases/22.03.7/packages/aarch64_cortex-a53/packages/openwisp-config_1.0.1-1_all.ipk](https://downloads.openwrt.org/releases/22.03.7/packages/aarch64_cortex-a53/packages/openwisp-config_1.0.1-1_all.ipk) in tmp# **curl https://downloads.openwrt.org/releases/22.03.7/packages/aarch64_cortex-a53/packages/openwisp-config_1.0.1-1_all.ipk -o openwisp-config_1.0.1-1_all.ipk** and then **opkg install openwisp-config_1.0.1-1_all.ipk --force-downgrade** and immediately it registered.

diepeterpan, 2024-10-19:
Hunting down the bug I discovered that **/etc/init.d/openwisp_config** wants to execute _PROG="**/usr/sbin/openwisp-config**"_ but it does not exists, it is called _PROG="**/usr/sbin/openwisp_config**"_ according to the changelog there were changes re: the - 'hyphen' vs _ 'underscore' ; maybe it just gets packaged incorrectly?  Once I edited **/etc/init.d/openwisp_config** and changed the - 'hyphen' to a  _ 'underscore' in _PROG=_ the new agent also started to work. Easier than rolling back.

diepeterpan, 2024-10-19:
Looks like the problem is in the OpenWrt Makefile - [https://github.com/openwrt/packages/blob/dd2e7105e6f0abf9b72155e2b26298ed19a92ab9/admin/openwisp-config/Makefile](https://github.com/openwrt/packages/blob/dd2e7105e6f0abf9b72155e2b26298ed19a92ab9/admin/openwisp-config/Makefile) which needs to be updated to produce hyphens instead of underscores.

```
	$(INSTALL_BIN) \
		$(PKG_BUILD_DIR)/openwisp-config/files/openwisp.agent \
		$(1)/usr/sbin/openwisp_config

	$(INSTALL_BIN) \
		$(PKG_BUILD_DIR)/openwisp-config/files/openwisp.init \
		$(1)/etc/init.d/openwisp_config
```

nemesifier, 2024-10-23:
https://github.com/openwrt/packages/pull/25186

closed 2024-12-13 by nemesifier